ArduPilot/ardupilot #30958: Caddx GM3 does not work with firmware > 3.4

twistedwings, 2025-08-21:
Caddx GM3 works fine with gimbal firmware 3.4.
With firmware 3.6 or 3.8 it does not work.

**Version**
ArduCopter 4.6.2

**Platform**
[  ] All
[  ] AntennaTracker
[ X] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Copter

**Hardware type**
Matek H743 Slim

GurmannSinghB, 2025-08-21:
i have tested GM1 and i have noticed that it works fine with firmware version 3.4 but facing difficulty with higher version.
basically GM1 stays retracted but doesn't respond to the mavlink msg.

GM1
ArduCopter 4.6.2
stm32F405

IamPete1, 2025-08-21:
I have a GM3 working on 4.7.dev. Make sure your on the latest gimbal firmware.

4O6FPV, 2025-08-31:
> I have a GM3 working on 4.7.dev. Make sure your on the latest gimbal firmware.

they are talking about caddx firmware above 3.4.

twistedwings, 2025-09-25:
I can confirm it works with gimbal firmware 3.8 after setting all channels to CH01.

closed 2025-10-05 by IamPete1

rmackay9, 2025-10-06:
yes and the wiki has been updated as well (see "Configuring the Gimbal" https://ardupilot.org/copter/docs/common-caddx-gimbal.html#configuring-the-gimbal)