ArduPilot/ardupilot #30975: Temperature sensor missing from H743 build

LupusTheCanine, 2025-08-22:
## Bug report

**Issue details**
TEMP* parameters are missing from the release build even though AP_TEMPERATURE_SENSOR_ENABLED is listed in features.txt

**Version**
4.6.2
**Platform**
[ ? ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
N/A

**Hardware type**
H743

**Logs**
N/A

JWaik, 2025-09-17:
I found that the problem has been fixed in the latest dev version with [this commit](https://github.com/ArduPilot/ardupilot/commit/6338d10b5fc817b26a61c9959aa7c6ec0389db0a#diff-85564705aaabd4078d2bdcbd56ca87fc94a95e3ff47b626b062fa80da0889423). However, the fix depends on using a board with external flash storage.

peterbarker, 2025-11-10:
So the story here is that unfortunately AP_TEMPERATURE_SENSOR_ENABLED is lying about whether the feature is built in.

What's being detected there is a set of "dummy methods" we use to have the temperature sensor code compiled in by default on Sub but not on any other vehicle.

So it's actually not a matter of "TEMP*" parameters missing - it really isn't compiled into the code, regardless of what features.txt says.

You can use the custom build server to actually compile it in.

As @JWaik found, we *do* compile it in on boards with  more than 2MB of flash.

LupusTheCanine, 2025-11-10:
> You can use the custom build server to actually compile it in.

I did that, it would be nice if it wasn't lying.

rob-clarke, 2025-11-26:
Have been exploring this myself. For boards between 1MB and 2MB flash (and not Sub), if `AP_TEMPERATURE_SENSOR_ENABLED` is not explicitly set in the hwdef, the current setup results in a strange situation where the standard builds don't have any real temp sensor support, but building on the custom build server with only the feature defaults set does include the support. My expectation was that building with the feature defaults would get the same result as the non-custom build.

A potential fix (though brittle) for the lying feature detection would be to use an `ExtractFeatures.FindString` in `extract_features.py` to look for 'AP_TemperatureSensor must be singleton' to detect if the real or dummy implementation has been used.

A nicer approach might be to use the symbol size (that's captured but not used as far as I can see) but I'm not sure if that is more prone to breaking. Is there a reliable correlation between code and symbol size?