ArduPilot/ardupilot #31105: uBlox M10 / F10 not respecting navfilter and constellation settings

wysiwyng, 2025-09-16:
## Bug report

**Issue details**

Parameters `GPS_NAVFILTER` and `GPSx_GNSS_MODE` have no effect on uBlox Gen 10 (tested with M10 and F10 chipsets, Sequre M10-25Q, HGLRC M100-5883 and Micoair MG-F10 boards). Skimming over the relevant code, this seems to have been introduced by #22248, where said parameters are overridden by hardcoded values.

The immediate effect of this is not being able to set said parameters to the desired values other than using u-center and saving the configuration to flash. Also if those parameters are set to anything else than the hardcoded values, a pre-arming check "GPS 1 still configuring this GPS" fails and prevents arming.

Relevant code lines:
- M10 hardcoded settings:
https://github.com/ArduPilot/ardupilot/blob/df83545f0b17c0c66425b3f6162a5c5946fbeaff/libraries/AP_GPS/AP_GPS_UBLOX.cpp#L245-L254
https://github.com/ArduPilot/ardupilot/blob/df83545f0b17c0c66425b3f6162a5c5946fbeaff/libraries/AP_GPS/AP_GPS_UBLOX.cpp#L496-L506
- Navfilter settings based on AP configuration:
https://github.com/ArduPilot/ardupilot/blob/df83545f0b17c0c66425b3f6162a5c5946fbeaff/libraries/AP_GPS/AP_GPS_UBLOX.cpp#L1143-L1157

**Version**
4.6.2

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tested on flying wing, not relevant

**Hardware type**
SpeedyBee F405 Wing APP

**Logs**
```
GPS 1 still configuring this GPS
GPS 1: u-blox M10 configuration 0x10000
```

IamPete1, 2025-09-16:
If you save a config with u-center you need to turn off auto config to prevent the conflict.

wysiwyng, 2025-09-17:
The issue also occurs when any of the 3 modules have their flash / BBR configuration cleared.

Also I would expect that no conflict should occur when settings are indeed present in flash memory, assuming ArduPilot sends the configuration settings in RAM take precedence over those in flash storage.

peterbarker, 2025-09-30:
@yuri-rage @bugobliterator we should take this into account with our refactoring of the val_set/val_get configuration system.

We're simply ignoring the user's parameters by using this static list of settings for the M10 (and it's unclear how we avoid that while keeping the "make it not suck" reasons we're disabling glonass).  No idea why we're hard-specifying the filter/model type used.

closed 2026-02-26 by peterbarker