ArduPilot/ardupilot #31108: Watchdog Reset Caused by Slow SD Card or Removal (Possibly Intermittent connection) During Flight

aero11223344, 2025-09-17:
**## Bug report**
When the SD card becomes slow or stops logging during flight, or on ground, removed and reinserted while the vehicle is armed, the system reports:

```
AP_Logger: Stuck Thread (Write)
Internal Error: 0x800
```

Following these errors, the CPU load increases sharply (~70% above) triggering a watchdog reset. This can be seen in the log as:

`WDG: T6 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0 I`

This causes the flight controller to reboot, which results in a loss of control and a crash if the aircraft is airborne.

The attached log is a sample of what has been seen. Here the bug is simulated by frequently inserting and removing the card and the error msg with watchdog reset activity appears.

**Version**
4.0.5, 4.4.4, 4.5.5, 4.6.2

**Platform**
[  ] Copter

**Airframe type**
hexarotor

**Hardware type**
Pixhawk4

**Logs**
https://drive.google.com/file/d/1rkvh2qtm10GrCSpO920fzp4OTniBobGF/view?usp=drive_link

tpwrules, 2025-11-11:
I have a Pixhawk 4 Mini and a Pixhawk 5X now and I am unable to replicate this issue on either.

tpwrules, 2025-11-12:
Tracked this down to a bug in the ChibiOS SD card driver for F4/F7. Confirmed that bug can't affect H7.

Working with ChibiOS on a solution.

tridge, 2025-12-01:
fixed in #31620