ArduPilot/ardupilot #31124: SRC_USERx unable to enable for use

Datto1200, 2025-09-18:
## Bug report

**Issue details**
SRC_USERx is listed in the full parameter list however it in not able to be enabled for use in the OSD options for display of custom .LUA messages

_Please describe the problem_
The full parameter list does let you change the value however the option tab is empty and does not enable for use

**Version**

4.6.2 and also 4.7.0-DEV

**Platform**

[  ] Copter

**Airframe type**

Quad "Frame type H"

**Hardware type**

MicroAir H743v2

**Logs**
No logs available

IamPete1, 2025-09-19:
Can you provide more details? What behavior are you expecting? What behavior did you actually get?

Datto1200, 2025-09-19:
from my understanding the function of SCR_USERx is for custom .LUA messages to be displayed via the OSD,
my usage is that I am running .LUA to have control of a thermal imaging camera colour palette selection via switch input on the controller and want the .LUA to output the name of the palette selected for 3 seconds on OSD then disappear.
In the full parameter list i can see that SCR_USER is available however it can not be enabled so that i can place it in the correct location on the OSD editor. at the moment I have .LUA messages redirected to OSDx_MESSAGE

closed 2025-09-19 by IamPete1