ArduPilot/ardupilot #31129: Bad value ID for INV3_ID_ICM42688

pitlab, 2025-09-19:
## Bug report

**Issue details**

I am making board with IMU ICM42688. There was problem with detecting this chip, because after reading WHO_AM_I register code check for incorrect value:
#define INV3_ID_ICM42688      0x47
It should be:
#define INV3_ID_ICM42688      0xDB 

**Version**
v4.7.0-dev (f3537)

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Not important

**Hardware type**
Custom board

**Logs**
No logs

andyp1per, 2025-09-20:
Please post a link to the reference manual where this value is used. The current value is correct because the vast majority of flight controllers use it and the value has no issues.

pitlab, 2025-09-20:
There are 2 version of ICM42688: with suffix -V and -P.
The ICM-42688-P has WHO_AM_I = 0x47 - so You have right
The ICM-42688-V has WHO_AM_I = 0xDB - it is just my case. I bought it in Mouser: https://product.tdk.com/system/files/dam/doc/product/sensor/mortion-inertial/imu/data_sheet/ds-000439-icm-42688-v.pdf?ref_disty=mouser

peterbarker, 2025-09-30:
@pitlab could you test https://github.com/ArduPilot/ardupilot/pull/31199 , please?

My reading seems to say they're the same part except the `-V` has extra stuff in it that we won't use, so the PR just makes us treat the -V and the -P