ArduPilot/ardupilot #31164: SequreH743 it doesn't load

Sekilsgs2, 2025-09-25:
I'm test this FW - https://firmware.ardupilot.org/Copter/stable-4.6.2/SequreH743/arducopter_with_bl.hex
And compiled it myself - it doesn't load
The first boot - you can see that the bootloader is loading, then it seems to have loaded, but it doesn't connect, and after about 20 seconds - it reboots itself and corrupts the firmware - you can then endlessly flash it through MissionPlanner, but it will never load.

peterbarker, 2025-09-30:
I'm unable to reproduce this locally.

This could be the H743-ECC-corruption problem which I think we have mitigations for with the bootloader from the master branch.  If you want to run stable I suggest you flash the "latest" firmware using DFU then flash the stable version using the bootloader from "latest".

If problems re-occur after doing that please reopen this issue with a comment saying that didn't work :-)

closed 2025-09-30 by peterbarker

andyp1per, 2025-09-30:
The newer boards require a patch that is not in 4.6.2. Here is a build with that patch and with the bootloader updated - please try it:

[arducopter_with_bl.zip](https://github.com/user-attachments/files/22614210/arducopter_with_bl.zip)