ArduPilot/ardupilot #31182: tridge TODO

tridge, 2025-09-29:
some things I've promised I would implement, done as public TODO

- 32 bit system ID for mavlink2 https://github.com/mavlink/rfcs/pull/20
- full 32 bit integers in PARAM_SET using new enum value
- drag coefficient values for copters in EKF3 using iterative EKF3 replay on replay log

shbhmexe, 2025-10-26:
Hi, could you please assign this issue to me? I’ve reviewed the project’s Code of Conduct and Contribution Guidelines, and I have the required skills and understanding of the tech stack to work on this effectively. I’ve already started analyzing the codebase and begun initial work toward resolving the issue. I’m confident I can implement a clean, optimized, and maintainable solution that aligns with the project’s standards. Looking forward to contributing!

tpwrules, 2025-11-06:
from pymavlink PR 1143: julian oes " I want to do param v3, but ArduPilot doesn't want it, from what I hear."