ArduPilot/ardupilot #31184: Documented parameter PSC_ACC_XY_FILT does not exist anymore

Georacer, 2025-09-29:
## Bug report

**Issue details**

We document the parameter `PSC_ACC_XY_FILT` [in the code](https://github.com/Georacer/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_PosControl.cpp#L85) and [in the wiki](https://ardupilot.org/copter/docs/parameters-Copter-stable-V4.6.2.html#psc-acc-xy-filt-xy-acceleration-filter-cutoff-frequency).
However, this parameter doesn't seem to actually exist anymore.

**Version**
4.6.2 and master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Simple multicopter, but probably all frames are affected.

**Hardware type**
N/A

**Logs**
N/A

Georacer, 2025-09-29:
@rmackay9 you mean we leave the old parameters in the new firmware, because MP always uses the new params XML regardless of the AP version connected?

Kind of makes sense:
1. Our tuning guides will mention only the correct parameters.
2. MP will generally present all the information needed (given its technical drawbacks).
3. If someone (like me) is weird enough to want to parse all the parameters from the Complete List and understand more about their usage... then that's a small enough number of confused users.

peterbarker, 2025-09-30:
> [@peterbarker](https://github.com/peterbarker) Any ideas whether it's possible to have an autotest for this type of issue?

We can, but while we're requiring old parameters to be present to make GCSs work we can't :-(

peterbarker, 2025-09-30:
I think we should remove it - the commit @MichelleRos  tracked down is from 2021!