ArduPilot/ardupilot #31194: AP_Logger: When using flash storage, a parameter change is required from 4.6 to 4.7

pkocmoud, 2025-09-29:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

When bringing up a new flight controller which has a W25Nxx 1Gb onboard SPI flash chip for logging. I built and tested both 4.6.2 (backported) and 4.7 copter firmware. In 4.6.2, I needed to set LOG_BACKEND_TYPE = 4 while in 4.7, I needed to change it to LOG_BACKEND_TYPE = 1. Both logs seem fine if this is set accordingly. In 4.7 with LOG_BACKEND_TYPE = 4, I see files created in the DataFlash but they are all (1.99Kb) and unusable. In 4.6 with LOG_BACKEND_TYPE = 1, I see no files listed.

In the HWDEF it is enabled in both builds by:

define HAL_LOGGING_DATAFLASH_ENABLED 1
define HAL_LOGGING_DATAFLASH_DRIVER AP_Logger_W25NXX

DATAFLASH littlefs:w25nxx

**Version**
4.6.2, 4.7.0-dev (f5fcafcb)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
X4 Copter

**Hardware type**
HCR-565 Flight Controller

**Logs**
NA

tpwrules, 2025-09-30:
Yes, you would need to change it since the littlefs acts as a filesystem so you need the file logging backend. I don't recall a parameter conversion.

But I am confused why both dataflash and littlefs are enabled, they should be mutually exclusive as the flash format is completely different. The board mentioned is not in tree, but it seems we do it on at least one (AEDROXH7). If only littlefs is enabled that may provide a slightly more clear problem as logging should fail completely. But yeah we probably do need to automatically switch. I don't know which boards we will do littlefs on for 4.7 in any case...

@andyp1per thoughts?

pkocmoud, 2025-09-30:
I used the board you referenced to work out how to support the flash logging. I just submitted a PR for my new board here: https://github.com/ArduPilot/ardupilot/pull/31200 using the 4.7 parameters.

andyp1per, 2025-09-30:
> Otherwise - We are currently flying with 4.6.2 and I'm hoping to get it included in 4.6.3, if at all possible. That said, I do not have a clear way to commit a 4.6.3 and a 4.7 compatible HWDEF. Please let me know how to proceed.

This is always true. @peterbarker changes lots of defines between releases so it will never be the case that you can have one size fits all

tpwrules, 2025-09-30:
> OK, so I think we have a confirmed problem here for 4.7 - we really need a smooth upgrade path for users of the affected boards.

https://github.com/ArduPilot/ardupilot/issues/31194#issuecomment-3351568187

Is this sufficient? We should update the default. This user might have been affected since they accidentally enable both.

andyp1per, 2025-09-30:
> @andyp1per,  do you suggest I commit the version that works with 4.6.3 and not 4.7 in the hopes that it can be included in the next release?

Do the PR for 4.7 then once merged do a back port PRAndySent from my iPhone

tpwrules, 2025-10-05:
This should prevent accidentally enabling both in the future: https://github.com/ArduPilot/ardupilot/pull/31236

tpwrules, 2025-10-08:
That PR has been merged so this should now cause an error.

I tested the upgrade behavior on KakuteH7Mini-Nand:
1. Flash stable 4.6.2 copter, erase params and dataflash
2. `LOG_BACKEND_TYPE` is default 4, make sure logs work (they do)
3. Flash master, don't erase params or dataflash
4. `LOG_BACKEND_TYPE` is default 1, make sure logs work (they still do but old logs have been deleted of course)

@peterbarker is that sufficient for user upgrades? Do we need a parameter conversion? 99.9999% of people will not touch that parameter. Unfortunately it looks like having the wrong `LOG_BACKEND_TYPE` may not cause arming failure.

If you don't believe we need to change anything then I think this issue can be closed. It was caused by accidentally enabling both block logging and LittleFS which does not work and will now raise an error.

peterbarker, 2025-11-01:
> [@peterbarker](https://github.com/peterbarker) is that sufficient for user upgrades? Do we need a parameter conversion? 99.9999% of people will not touch that parameter. Unfortunately it looks like having the wrong `LOG_BACKEND_TYPE` may not cause arming failure.

I think it would be OK if we *did* get an arming failure.  And we should as we have arming checks that logging is working.

But only because, as you say, the vast majority of people won't have changed from the default.

Others may disagree on DevCall with this assessment.